simplefoc/Arduino-FOC-drivers #1: BitOrder is not defined in all plaforms

jkirsons, 2021-04-29:
https://github.com/simplefoc/Ardunio-FOC-drivers/blob/fc74bae6a996e1c556d79dee22d4386bcd85175f/src/encoders/as5048a/AS5048A.h#L52

BitOrder is not defined in all plaforms.  See: https://github.com/adafruit/Adafruit_BusIO/blob/13bd6906a62b5f2e8cbe8529f55ad13880ffd730/Adafruit_SPIDevice.h#L6

Currently this does not compile on the ESP32 platform.

In file included from .pio\libdeps\pico32\Simple FOC drivers\src\encoders\as5048a\AS5048A.cpp:8:0:
.pio\libdeps\pico32\Simple FOC drivers\src\encoders\as5048a\AS5048A.h:52:47: error: 'BitOrder' has not been declared
 static SPISettings AS5048SPISettings(1000000, BitOrder::MSBFIRST, SPI_MODE1);
                                               ^

runger1101001, 2021-04-29:
Thanks, I'll replace it by its numeric value or see if there's a #define like for SPI_MODE1

jkirsons, 2021-05-10:
Hi, MSBFIRST and LSBFIRST are already #define's, so they cannot be used again in the enum declaration.

Are you able to use MSBFIRST/LSBFIRST directly, without the BitOrder enum?
Eg, this will compile for me without any ESP32 specific code added:

`static SPISettings AS5048SPISettings(8000000, MSBFIRST, SPI_MODE1);`

runger1101001, 2023-04-14:
This was a big pain, but I believe it is now fixed and compiling clean for all our platforms. I will close this ticket, please re-open if you still experience the issue.

closed 2023-04-14 by runger1101001