simplefoc/Arduino-FOC-drivers #2: Issues when installing this library

steph643, 2021-06-12:
1.
It seems the library cannot be installed through the Arduino IDE Library Manager. The doc says:
>Open Arduino IDE and start Arduino Library Manager by clicking: Tools > Manage Libraries....
>Search for Simple FOC drivers library and install the latest version.

But no "Simple FOC drivers" library can be found this way.

2.
When installing the library manually from Github, the doc says:
> Reopen Arduino IDE and you should have the library examples in File > Examples > Simple FOC drivers.

But the library doesn't appear in `File > Examples`. It appears in `Sketch > Include Library` though.

steph643, 2021-06-12:
Also, on [this page](https://github.com/simplefoc/Ardunio-FOC-drivers/tree/master/src/encoders/as5048a), the proposed code doesn't compile under Arduino IDE:
```
#include "SimpleFOC.h"
#include "encoders/as5048a/MagneticSensorAS5048A.h"
```
What seems to be required is this:
```
#include "SimpleFOC.h"
#include "SimpleFOCDrivers.h"
#include "encoders/as5048a/MagneticSensorAS5048A.h"
```

runger1101001, 2021-10-12:
fixed the second point in the docs.

regarding the first point, the library still has to be made "official"... fixing the documentation errors is a good first step so thank you very much for the feedback!

closed 2021-10-12 by runger1101001