simplefoc/Arduino-FOC-drivers #10: Support Infineon's TLE5012B SPI encoder

zhuangfengzi, 2022-02-03:
The accuracy of the tle5012b encoder is 16 bit or 15 bit, and the performance is very good! His SPI was originally 3 wires, and the official added 4-wire library, which is convenient to connect! Can add this encoder may give me some advice on code transformation! esteem it a favor!

https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor

runger1101001, 2022-02-03:
There is support for the TLE5012B for STM32 now: 
https://github.com/simplefoc/Arduino-FOC-drivers/tree/master/src/encoders/tle5012b

And there is a new GenericSensor class:
https://docs.simplefoc.com/generic_sensor
with which you should be able to integrate the official library and just use `Tle5012MagneticAngleSensor.getAngleValue(d);` on the hardware platforms it supports.

runger1101001, 2023-04-14:
Since we now support the TLE5012B, at least on STM32 MCUs, I am going to close this ticket. Please feel free to re-open it or create a new one if the solution we have isn't meeting your needs.

closed 2023-04-14 by runger1101001