simplefoc/Arduino-FOC-drivers #27: question

eduardsik, 2023-08-04:
I try to initialize AS5047A on a RP2040, default SPI port with Cs on 22 pin. No success. 

`

> #include <Arduino.h>
> #include <Wire.h>
> #include <SPI.h>
> #include <SimpleFOC.h>
> #include <encoders/as5048a/MagneticSensorAS5048A.h>
> 
> #define SENSOR1_CS 5 // some digital pin that you're using as the nCS pin
> MagneticSensorAS5048A sensor1(SENSOR1_CS);
> 
> void setup() {
> 
>   Serial.begin(115200);
>   // initialize magnetic sensor hardware
>   sensor1.init();
> 
>   Serial.println("sensor ready");
>   _delay(1000);
> }
> 
> void loop() {
>  // IMPORTANT - call as frequently as possible
>   // update the sensor values 
>   sensor1.update();
>   
>   // get the angle, in radians, including full rotations
>   float a1 = sensor1.getAngle();
>   Serial.print(a1);
>   Serial.print("  ");
>   // get the velocity, in rad/s - note: you have to call getAngle() on a regular basis for it to work
>   float v1 = sensor1.getVelocity();
>   Serial.println(v1);
> }

`

using platformio

Tested the sensor on arduino uno, and it is working fine. 
What am I not noticing?

Thanks.

VIPQualityPost, 2023-09-14:
which core are you using? mbed or hillpower?
there are some issues related to this on the forum as well

runger1101001, 2023-09-17:
May I ask which RP2040 board it is?

eduardsik, 2023-09-20:
Nevermind, I manage it to work. 

Here is a working code with open loop velocity control. 

`/**
 *
 * Velocity motion control example
 * Steps:
 * 1) Configure the motor and magnetic sensor
 * 2) Run the code
 * 3) Set the target velocity (in radians per second) from serial terminal
 *
 *
 * By using the serial terminal set the velocity value you want to motor to obtain
 *
 */
#include "Arduino.h"
#include "SPI.h"
#include "SimpleFOC.h"
#include "SimpleFOCDrivers.h"
#include "encoders/as5048a/MagneticSensorAS5048A.h"

#define SPI_2_MISO  12
#define SPI_2_MOSI  15
#define SPI_2_SCK   14
#define SENSOR_CS  13 // some digital pin that you're using as the nCS pin

MagneticSensorAS5048A sensor(SENSOR_CS, true);
MbedSPI spi2(SPI_2_MISO, SPI_2_MOSI, SPI_2_SCK);

// BLDC motor & driver instance
BLDCMotor motor = BLDCMotor(11);
BLDCDriver3PWM driver = BLDCDriver3PWM(9, 8, 7, 6);

// velocity set point variable
float target_velocity = 0;
// instantiate the commander
Commander command = Commander(Serial);
void doTarget(char* cmd) { command.scalar(&target_velocity, cmd); }

void setup() {
  spi2.begin();
  // initialise magnetic sensor hardware
  sensor.init(&spi2);
  // link the motor to the sensor
  motor.linkSensor(&sensor);
  // pwm frequency to be used [Hz]
  driver.pwm_frequency = 200000;
  // power supply voltage
  driver.voltage_power_supply = 12;
  driver.init();
  motor.linkDriver(&driver);

  // limiting motor movements
  motor.phase_resistance = 5.3; // [Ohm]
  motor.current_limit = 2;   // [Amps] - if phase resistance defined
  motor.voltage_limit = 3;   // [V] - if phase resistance not defined
  motor.velocity_limit = 5; // [rad/s] cca 50rpm

  // set motion control loop to be used
  motor.controller = MotionControlType::velocity;

  // contoller configuration
  // default parameters in defaults.h

  // velocity PI controller parameters
  motor.PID_velocity.P = 0.5f;  //0.4
  motor.PID_velocity.I = 20;  //20
  motor.PID_velocity.D = 0; //0
  // default voltage_power_supply
  motor.voltage_limit = 12;
  // jerk control using voltage voltage ramp
  // default value is 300 volts per sec  ~ 0.3V per millisecond
  motor.PID_velocity.output_ramp = 1000;

  // velocity low pass filtering
  // default 5ms - try different values to see what is the best.
  // the lower the less filtered
  motor.LPF_velocity.Tf = 0.01f;

  // use monitoring with serial
  Serial.begin(115200);
  // comment out if not needed
  motor.useMonitoring(Serial);

  // initialize motor
  motor.init();
  // align sensor and start FOC
  motor.initFOC();

  // add target command T
  command.add('T', doTarget, "target velocity");

  Serial.println(F("Motor ready."));
  Serial.println(F("Set the target velocity using serial terminal:"));
  _delay(1000);
}

void loop() {
  // main FOC algorithm function
  // the faster you run this function the better
  // Arduino UNO loop  ~1kHz
  // Bluepill loop ~10kHz
  motor.loopFOC();

  // Motion control function
  // velocity, position or voltage (defined in motor.controller)
  // this function can be run at much lower frequency than loopFOC() function
  // You can also use motor.move() and set the motor.target in the code
  motor.move(target_velocity); 
  //motor.move(10); 

  // function intended to be used with serial plotter to monitor motor variables
  // significantly slowing the execution down!!!!
  // motor.monitor();

  // user communication
  command.run();
}
`

runger1101001, 2023-09-21:
Great!

closed 2023-09-21 by runger1101001