simplefoc/Arduino-FOC-drivers #31: MT6835 Arduino Example Code

GvnniL, 2023-11-01:
I have followed the procedure mentioned in the MT6835 README (and modify the nCS to Arduino pin 10, SimpleFOCDebug::print to Serial.print) but the output whether angle or speed stays around 0 or 12861 when I tried to rotate the motor shaft with a magnet.
```
A: 0.06
A: 0.07
A: 0.00
A: 12861.62
A: 12861.61
A: 0.04
A: 0.04
A: 12861.59
A: 12861.66
A: 0.03
A: 0.01
A: 0.07
A: 0.06
A: 12861.61
A: 12861.61
A: 12861.61

```

Prelo96, 2025-07-25:
Same here, no data at all, I also tried the programming sketch but only wrong data from request and crash whet set data

runger1101001, 2025-07-27:
I have successfully used this driver, as have other people. So in order to find out what is not working you’ll have to share more details about your setup, MCU type and the code you’re trying to run…

itaistrauss-eng, 2025-08-06:
Hi. I received an MT6835 chip ( [aliexpress](https://www.aliexpress.com/item/1005009175953978.html?channel=twinner) ) and connected it to an Arduino and set it up using the example code ( [example setup](https://github.com/simplefoc/Arduino-FOC-drivers/tree/master/src/encoders/mt6835#software-setup) ). I also encountered 'range' problem with encoder getCurrentAngle() function. 
I switched to the getCurrentAngle() function and find out that the data only has a range 16 bits instead of 21 bits. that translate into about -5.625 to +5.625 degrees (or -32767 to 32767 counts per revolution (i.e. signed 16 bits).
I haven't yet calibrated the chip, but that should not be an issue (yet). And I relying on the [Arduino-FOC-drivers](https://github.com/simplefoc/Arduino-FOC-drivers/tree/master) library for translate the raw data into angle.

Does anyone else encountered this issue? or found a solution ?

itaistrauss-eng, 2025-08-09:
I have an update. I succeeded reading the full 21bit only after applying the readRegister directly to every angle relevant address: 0x003, 0x004, 0x005 and 0x006. I don't know why the burst command doesn't work. Maybe its a bug in the library code or maybe its a bug in the chip itself.
Using the burst mode yield a 16 bit number with bits 16 to 20 copying themseld from bit 15. And the CRC check was ok, so the coding could be wrong at the chip level.
For successful direct readout of the registers I had to delete the private declaration in MT6835.h 

Would love to know if anyone had a different solution