simplefoc/Arduino-FOC-drivers #37: [FEATURE] Allowing to choose SPI pins

sylque, 2024-02-02:
_This is a duplicate of https://github.com/simplefoc/Arduino-FOC/issues/375_

**Is your feature request related to a problem? Please describe.**
On my ESP32, I can choose any GPIO pins for SPI. Unfortunately, it seems SimpleFOCDrivers only allows for the two sets of default SPI pins (HSPI and VSPI), like this:
```c++
static SPIClass spi(HSPI);
sensor.init(&spi);
```

**Describe the solution you'd like**
```c++
sensor.init(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);
```

runger1101001, 2024-02-02:
Hi there,

I think you can already do this:

```
    SPIClass mySPI(HSPI);
    mySPI.begin(PIN_CLK, PIN_MISO, PIN_MOSI);
    sensor.init(&mySPI);
```

Please let me know if this solves your problem?

sylque, 2024-02-02:
Yes, this should work! I will post the result here.

sylque, 2024-02-02:
It works perfectly, thank you very much.

Now I have 3 places with SPI settings instead of two :-) (sensor constructor, sensor.init(), spi.begin())

closed 2024-02-02 by sylque