simplefoc/Arduino-FOC-drivers #38: Question: fastMode of the AS5048A

sylque, 2024-02-06:
Hi Runger,
Would you please explain when the fastMode should be used? What's the benefit of calling readRawAngle() twice?
Thanks,

runger1101001, 2024-02-14:
Dear sylque,

You can use the fast mode if you want - the difference is the following:

- without fast mode, the sensor is read twice - the second angle read is the current angle, which the result of the first read-angle command
- with fast mode, it reads the sensor only once - this means the angle read is the result of the previous read-angle command, from the last iteration. 
 
So in fast mode, the SPI reads are 2x faster, but the angle result is one iteration old - there is more lag between the angle and the real rotor position.

Which is better really depends on your application, normally I would say not to use fast mode is probably better.

sylque, 2024-02-15:
Thanks a lot for the explanation. 

I've just had a look at my SPI implementation (ESP32): I'm no expert, but I don't see anything indicating that `spi->transfer16()` does any buffering. It just seems to wait until it gets the most recent data. So until more tests are done, I will stick to fastMode=true :-)

Thanks again.

closed 2024-02-15 by sylque